StanfordVL/BEHAVIOR-1K #805: Run examples.objects.draw_bounding_box in og-develop branch failed

hoarjour, 2024-07-17:
Firstly，I got error message at line 69: obs, _ = cam.get_obs()

> Traceback (most recent call last):
  File "D:\miniconda3\envs\omni\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\miniconda3\envs\omni\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\examples\objects\draw_bounding_box.py", line 87, in <module>
    main()
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\examples\objects\draw_bounding_box.py", line 69, in main
    obs, _ = cam.get_obs()
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\sensors\sensor_base.py", line 76, in get_obs
    obs, info = self._get_obs()
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\sensors\vision_sensor.py", line 284, in _get_obs
    obs[modality] = self._remap_bounding_box_semantic_ids(obs[modality])
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\sensors\vision_sensor.py", line 411, in _remap_bounding_box_semantic_ids
    bbox["semanticId"] = VisionSensor.SEMANTIC_REMAPPER.remap_bbox(bbox["semanticId"])
  File "D:\PycharmProjects\OmniGibson-main\omnigibson\utils\vision_utils.py", line 140, in remap_bbox
    assert semantic_id < len(self.key_array), f"Semantic id {semantic_id} is out of range!"
AssertionError: Semantic id 0 is out of range!

I found #690 has the same problem, then I tried to run the code in og-develop branch, but got error message:

> Traceback (most recent call last):
  File "D:\miniconda3\envs\omni\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\miniconda3\envs\omni\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\examples\objects\draw_bounding_box.py", line 89, in <module>
    main()
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\examples\objects\draw_bounding_box.py", line 69, in main
    obs, _ = cam.get_obs()
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\sensors\sensor_base.py", line 78, in get_obs
    obs, info = self._get_obs()
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\sensors\vision_sensor.py", line 307, in _get_obs
    obs[modality] = self._remap_bounding_box_semantic_ids(obs[modality])
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\sensors\vision_sensor.py", line 462, in _remap_bounding_box_semantic_ids
    bbox["semanticId"] = VisionSensor.SEMANTIC_REMAPPER.remap_bbox(bbox["semanticId"], self._scene)
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\utils\vision_utils.py", line 156, in remap_bbox
    return semantic_class_name_to_id(scene)["unlabelled"]
  File "D:\new_PycharmProject\OmniGibson-og-develop\omnigibson\utils\constants.py", line 182, in semantic_class_name_to_id
    systems = sorted(scene.system_registry.object_names)
AttributeError: 'NoneType' object has no attribute 'system_registry'

I don't know how to solve this problem. Any solutions for this?

cgokmen, 2024-07-26:
We will get this fixed on og-develop asap. Will let you know.

hang-yin, 2024-07-26:
@hoarjour sorry for the wait, this should be fixed on `og-develop` now!

closed 2024-07-26 by hang-yin