StanfordVL/BEHAVIOR-1K #807: Container how to be filled with fluid?

SresserS, 2024-07-18:
Hi! I want to know how to fill the containers(like sink, bathtub or swimming pool ) with water ? I have tried water=get_system("water")
    assert sink.states[object_states.Saturated].set_value(water,True)
but it didn't work

ChengshuLi, 2024-07-18:
You can try ` sink.states[object_states.Filled].set_value(water, True)`.

SresserS, 2024-07-18:
Hi, Thanks @ChengshuLi ! I have tried this , but there are still some problems,  my code is as follows:
`bathtub_cfg = dict(
        type="DatasetObject",
        name="bathtub",
        category="bathtub",
        model="fdjykf",
        abilities={
            # "toggleable": {},
            # "saturable":{},
            "fillable":{},
        },
        position=[0.0, 0, 1.42],
    )

    cfg["objects"] = [bathtub_cfg]

    # Create the environment!
    env = og.Environment(configs=cfg)

    # Set camera to ideal angle for viewing objects
    og.sim.viewer_camera.set_position_orientation(
        position=np.array([ 0.37860532, -0.65396566,  1.4067066 ]),
        orientation=np.array([0.49909498, 0.15201752, 0.24857062, 0.81609284]),
    )

    # Take a few steps to let the objects settle, and then turn on the sink
    for _ in range(10):
        env.step(np.array([]))              # Empty action since no robots are in the scene

    sink = env.scene.object_registry("name", "bathtub")
    # assert sink.states[object_states.ToggledOn].set_value(True)
    water=get_system("water")
    og.log.info(sink.states.keys())
    sink.states[object_states.Filled].set_value(water,True)`

however , error is KeyError: <class 'omnigibson.object_states.filled.Filled'>
and the states dict only have `[INFO] [omnigibson] dict_keys([<class 'omnigibson.object_states.contact_bodies.ContactBodies'>, <class 'omnigibson.object_states.aabb.AABB'>, <class 'omnigibson.object_states.pose.Pose'>, <class 'omnigibson.object_states.saturated.ModifiedParticles'>, <class 'omnigibson.object_states.contact_particles.ContactParticles'>, <class 'omnigibson.object_states.saturated.Saturated'>, <class 'omnigibson.object_states.covered.Covered'>, <class 'omnigibson.object_states.adjacency.HorizontalAdjacency'>, <class 'omnigibson.object_states.touching.Touching'>, <class 'omnigibson.object_states.adjacency.VerticalAdjacency'>, <class 'omnigibson.object_states.particle_source_or_sink.ParticleSink'>, <class 'omnigibson.object_states.particle_source_or_sink.ParticleSource'>, <class 'omnigibson.object_states.next_to.NextTo'>, <class 'omnigibson.object_states.on_top.OnTop'>, <class 'omnigibson.object_states.under.Under'>, <class 'omnigibson.object_states.inside.Inside'>])
`

so can you inform me how can I fill a container with liquid?

SresserS, 2024-07-18:
oh i just tried a bowl , and it can be filled with water, so is there any other larger container have the filled state? and do they need some specific config setting to contain water?

SresserS, 2024-07-18:
Well , i find a swimming pool fillable , that's really nice! 😀

ChengshuLi, 2024-07-19:
They do need to contain a "fillable" volume metalink annotation for the `Filled` state to work properly.
Yes some swimming pools should be fillable, e.g. see the "Meta Links" in https://behavior.stanford.edu/knowledgebase/objects/swimming_pool-kohria/index.html.

SresserS, 2024-07-21:
> They do need to contain a "fillable" volume metalink annotation for the `Filled` state to work properly. Yes some swimming pools should be fillable, e.g. see the "Meta Links" in https://behavior.stanford.edu/knowledgebase/objects/swimming_pool-kohria/index.html.

Thanks @ChengshuLi , But maybe there is a bug that when the swimming pool is filled with water, at the beginning the water falls into the pool as if it were falling from the sky, collides, bounces, and finally comes to rest. I am wondering if there is a way to make the water still From beginning to end

SresserS, 2024-07-21:
[image]

ChengshuLi, 2024-07-29:
One thing you could do is:
1. set Filled to be True
2. let water falls and settles 
3. call `state = og.sim.dump_state()` to dump the state

Then you can do `og.sim.load_state(state)`, which should have the water still.