StanfordVL/BEHAVIOR-1K #808: Change projectionType for viewport to fisheyeSpherical

HSL31, 2024-07-18:
Hi, I've been trying to add in a fisheye camera to the scene and save rgb, depth, and segmentation results. I've been looking into the VisionSensor code and some other related ones and was not able to find where I can tangle with the projection type settings. 

Is there a certain code or config where we can set the type or do we have to directly import isaac-sim classes to add in a default camera and start from there. 

Thank you.

cgokmen, 2024-07-26:
We do not support this out of the box. If you can find a way to do it with raw Isaac Sim classes, please link to it here and we can help you integrate that into OmniGibson.

ChengshuLi, 2024-09-23:
Feel free to re-open the issue if the problem persists, thanks!