StanfordVL/BEHAVIOR-1K #818: Error in importing BoundingBoxAPI

krajaei, 2024-07-28:
Hi

I encountered an error related to import BoundingBoxAPI. The specific error message I received was as follows:
 
```

from omnigibson.utils.usd_utils import BoundingBoxAPI
ImportError: cannot import name 'BoundingBoxAPI' from 'omnigibson.utils.usd_utils' (/...OmniGibson/omnigibson/utils/usd_utils.py)

```
Upon examining the file “usd_utils.py”, I was unable to locate any function named BoundingBoxAPI. We recently installed Omnigibson and are using the latest version of omnigibson.
I would greatly appreciate your assistance in resolving this issues.

hang-yin, 2024-07-29:
Hi! 

Would you mind naming the specific branch you are working off of? As far as I know, we have deprecated the use of this `BoundingBoxAPI` since a while back. If you are on `og-develop` or `main`, this `from omnigibson.utils.usd_utils import BoundingBoxAPI` call should not exist anymore.

krajaei, 2024-07-29:
> Hi!
> 
> Would you mind naming the specific branch you are working off of? As far as I know, we have deprecated the use of this `BoundingBoxAPI` since a while back. If you are on `og-develop` or `main`, this `from omnigibson.utils.usd_utils import BoundingBoxAPI` call should not exist anymore.

Hi
If the use of 'BoundingBoxAPI' has been deprecated, can you please inform me of the alternative? For instance, what function could we use  in place of 'BoundingBoxAPI.aabb_contains_point()'? Thank you for your assistance.

hang-yin, 2024-07-30:
Hi @krajaei ! Sorry for the wait, here's the PR that removed the BoundingBoxAPI: https://github.com/StanfordVL/OmniGibson/pull/605

You can check the code changes to see how we got rid of it. Specifically, for aabb_contains_point(), you can follow this conversion:

`BoundingBoxAPI.aabb_contains_point(points, (container_aabb_lo, container_aabb_hi))`
->
`if not (np.less_equal(container_aabb_lo, points).all() and np.less_equal(points, container_aabb_hi).all())`

You can also reference the deprecated implementation of this method:
```
@classmethod
    def aabb_contains_point(cls, point, container):
        """
        Returns true if the point is contained in the container AABB
        Args:
            point (tuple): (x,y,z) position in world-coordinates
            container (tuple):
                - 3-array: start (x,y,z) corner of world-coordinate frame aligned bounding box
                - 3-array: end (x,y,z) corner of world-coordinate frame aligned bounding box
        Returns:
            bool: True if AABB contains @point, otherwise False
        """
        lower, upper = container
        return np.less_equal(lower, point).all() and np.less_equal(point, upper).all()
```

ChengshuLi, 2024-09-23:
Feel free to re-open the issue if the problem persists, thanks!

closed 2024-09-23 by ChengshuLi