StanfordVL/BEHAVIOR-1K #820: Does OmniGibson have some predefined high-level actions?

shengyin1224, 2024-07-30:
Hi guys, I am very excited for the OmniGibson simulator and want to try instruction following task in the platform. Therefore, I am wondering if there are some already created high-level actions like 'open', 'switchon' for interacting objects. I only find the action space of a type of robot is a vector to control the low-level motion.

-----------------------------------------------------------------------------------------

I  found 'symbolic_semantic_action_primitives' and 'starter_semantic_action_primitives' in the doc. I wanna know the difference between these two and it seems that the first one supports more high-level actions

hang-yin, 2024-07-31:
@cgokmen can best answer this.

ge79puv, 2024-10-09:
Hi, I have the same question. Any updates?

cgokmen, 2024-10-09:
You can use the action primitives which are now included in OmniGibson. You will need OMPL to use them so just start with the Docker container. There are examples in the examples/action_primitives directory, and you can find out more about the full set of primitives in the omnigibson/action_primitives directory. We have a physically simulated set and a symbolic, teleporting set.

closed 2024-10-09 by cgokmen

yananchen1989, 2025-03-28:
@cgokmen 

hi author, in omnigibson, are there any methods to create higher level actions, such as wipe, rinse, pour, clean etc ?

(I want the VLM/LLM-powered robot  to generate high-level actions with semantic meaning)

thanks.

shengyin1224, 2025-04-09:
> [@cgokmen](https://github.com/cgokmen)
> 
> hi author, in omnigibson, are there any methods to create higher level actions, such as wipe, rinse, pour, clean etc ?
> 
> (I want the VLM/LLM-powered robot to generate high-level actions with semantic meaning)
> 
> thanks.

They seem to provide some high-level actions in starter_semantic_action_primitives.py