StanfordVL/BEHAVIOR-1K #827: Unable to open the usd in isaac sim

Entongsu, 2024-08-01:
Hi,

I have tried to open the usd in the isaac sim but I got the error of 
```[Error] [omni.usd] Failed to open
```. I am wondering whehter there is any ways to open these usds in the isaac sim 4.1.0. Thank you.

ChengshuLi, 2024-08-02:
What USDs are you tryng to open?

Entongsu, 2024-08-02:
Hi,

I downloaded the data from the [link]("https://storage.googleapis.com/gibson_scenes/og_dataset_1_0_0.tar.gz) and tried to open the USD file in the Isaac sim 4.1.0, such as the sponge and got the error of ```Error] [omni.kit.window.file.file_window] Failed to open stage /home/lme/Downloads/og_dataset/objects/acorn/qkjrwt/usd/qkjrwt.encrypted.usd: Failed to open:/home/lme/Downloads/og_dataset/objects/acorn/qkjrwt/usd/qkjrwt.encrypted.usd
```.  Thank you!

ChengshuLi, 2024-08-02:
Thanks for replying! This is expected.

These USD files are encrypted because they can only be used within OmniGibson due to copyright reasons. You can use `DatasetObject(category="acorn", model="qkjrwt")` to load the object, which internally will decrypt the USDs for you. Let me know if you run into further issues.

closed 2024-08-02 by ChengshuLi

SresserS, 2024-09-30:
@ChengshuLi hi，I am wondering how can I modify your object like the axis, or how can I make a similar object with similar Properties? I really need a tool for robot.