StanfordVL/BEHAVIOR-1K #828: Simulation Is TOO slow

albertcity, 2024-08-02:
I am OmniGibson to doing research. However, I am quite concerned about the simulation speed of OmniGibson. It seams that OmniGibson is running quite slow (as shown in the following):
I checkout into the `og_develop` branch and running the `wip_solve_behavior_task.py`. I made the following changes to this file:
- set `load_task_relevant_only=False` to load an entire scene, which is essential for me;
- set `action_frequency=10, physics_frequency=30, gm.ENABLE_FLATCACHE=True` for a higher simulation speed;
- bypass the `StarterSematicActionPrimitives` (via sample a action from action_space instead of of from the action generator)

However, under above setting, the simulation speed of OmniGibson is just too slow; It can only call `env.step` 1.2 times per second, which means 1 seconds in simulation will takes 7.8 seconds to simulate in real-world. 

I have also tried to set the `device=cuda` in the `fetch_primitives.yaml`, however this seems not compatible with current codebase.

Is such a low speed is expected in OmniGibson? If not, how can I solve the problem?

ChengshuLi, 2024-08-02:
In that specific example, each `execute_controller` call actually involves multiple environment steps. Within each of these steps of primitive execution, motion planners as well as kinematic sampling code are being called, which contributes to the slow down. To clarify, the slow down is mainly due to the underlying motion planner and the kinematic sampling code (e.g. ray casting, collision checking, etc). Running OmniGibson's environment itself should be much faster (above real time). We are in the process of incorporating [cuRobo](https://curobo.org/) for motion palnning and kienmatic sampling, which alleviates the problem that you brought up. 

Hope this helps!

closed 2024-08-02 by ChengshuLi

albertcity, 2024-08-02:
> execute_controller
Thanks for your rely! Really looking forward to the upcoming updates :)