StanfordVL/BEHAVIOR-1K #832: Adjusting Scene Lighting

krajaei, 2024-08-07:
Hi 
Is there a way to adjust lighting in scenes?

hang-yin, 2024-08-12:
If you want to adjust lighting for all light sources in a scene, an easy way to do this is to adjust the `gm.FORCE_LIGHT_INTENSITY` constant in the `omnigibson/macros.py` file. Let me know if that works!

krajaei, 2024-08-14:
Thank you for your response. It works, although I've found that it requires adjusting before loading a scene. I was wondering if it is possible to dynamically adjust the lighting after the scene has been loaded?

hang-yin, 2024-08-15:
From our `omnigibson.utils.ui_utils`, there's a `CameraMover` class. You can try this:

1. grab the `viewer_camera` instance 
2. create a `CameraMover` with it
3. call `change_light` or `set_light` on the `CameraMover` object

ChengshuLi, 2024-09-23:
Feel free to re-open the issue if the problem persists, thanks!

closed 2024-09-23 by ChengshuLi