StanfordVL/BEHAVIOR-1K #848: Investigate / silence material errors

cgokmen, 2024-08-21:


shengyin1224, 2024-12-03:
Hi，I encouter the same issue. Have you already solved this problem?

VincentWanghh, 2024-12-30:
Hi, I also encouter this issue. Have you solved it? :)

ChengshuLi, 2024-12-31:
These errors are safe to ignore for now. The root cause is that we need to decrypt our encrypted assets on the fly and immediately delete them afterwards to comply with their copyright. By the time you see these errors, the texture maps are already loaded into memory and ready to be used by Isaac, even though the errors indicate Isaac can't find these texture maps.

Meanwhile, we understand these errors are rather confusing and misleading for our users. We will try to get rid of these in our next release.

cgokmen, 2025-02-12:
Concretely, this happens because even though we update all shader paths in post_load, sometimes we end up rendering objects before that, in which case they will render without a material. This gets corrected immediately after, but it spams the log stream and also adds some latency, so we should fix it still.