StanfordVL/BEHAVIOR-1K #865: download_datasets

fighting-j, 2024-09-04:
Hello, when I follow the installation steps to download the dataset, the following error appears. How can I solve it?

(omnigibson) ubuntu@hjj:~/OmniGibson$ python scripts/download_datasets.py
Requested average specs of the object categories in the OmniGibson Dataset of objects, but the file cannot be found. Did you download the dataset? Returning an empty dictionary
Traceback (most recent call last):
  File "/home/ubuntu/OmniGibson/scripts/download_datasets.py", line 7, in <module>
    from omnigibson.macros import gm
  File "/home/ubuntu/OmniGibson/omnigibson/__init__.py", line 11, in <module>
    from omnigibson.envs import Environment
  File "/home/ubuntu/OmniGibson/omnigibson/envs/__init__.py", line 1, in <module>
    from omnigibson.envs.env_base import Environment
  File "/home/ubuntu/OmniGibson/omnigibson/envs/env_base.py", line 6, in <module>
    from omnigibson.objects import REGISTERED_OBJECTS
  File "/home/ubuntu/OmniGibson/omnigibson/objects/__init__.py", line 1, in <module>
    from omnigibson.objects.object_base import REGISTERED_OBJECTS, BaseObject
  File "/home/ubuntu/OmniGibson/omnigibson/objects/object_base.py", line 11, in <module>
    from omnigibson.prims.entity_prim import EntityPrim
  File "/home/ubuntu/OmniGibson/omnigibson/prims/__init__.py", line 2, in <module>
    from omnigibson.prims.cloth_prim import ClothPrim
  File "/home/ubuntu/OmniGibson/omnigibson/prims/cloth_prim.py", line 14, in <module>
    from omnigibson.systems import get_system
  File "/home/ubuntu/OmniGibson/omnigibson/systems/__init__.py", line 4, in <module>
    from omnigibson.systems.micro_particle_system import *
  File "/home/ubuntu/OmniGibson/omnigibson/systems/micro_particle_system.py", line 21, in <module>
    import pymeshlab
  File "/home/ubuntu/anaconda3/envs/omnigibson/lib/python3.10/site-packages/pymeshlab/__init__.py", line 11, in <module>
    from .pmeshlab import *
ImportError: /home/ubuntu/anaconda3/envs/omnigibson/lib/python3.10/site-packages/pymeshlab/lib/libmeshlab-common.so: undefined symbol: _ZdlPvm, version Qt_5

hang-yin, 2024-09-10:
This is interesting, seems like a pymeshlab installation issue. This looks relevant: https://github.com/cnr-isti-vclab/PyMeshLab/issues/341
Do you mind sending your ubuntu version?

fighting-j, 2024-09-11:
[image]
My ubuntu version is 22.04.

ChengshuLi, 2024-09-23:
Can you try to just do 
```
import pymeshlab
```
in your conda environment?

Maybe try to reinstall it again?

hang-yin, 2024-09-23:
From @cgokmen: This could be a Qt version issue. It seems like pymeshlab expects one version, but you have another one installed.

ViperHFormer, 2024-09-27:
I wonder how to solve this problem