StanfordVL/BEHAVIOR-1K #879: ModuleNotFoundError: No module named 'omni.kit.usd'

Marxlp, 2024-09-13:
## Description

After install OmniGibson from souce, and start a example by running 

> **python -m omnigibson.examples.scenes.scene_selector**

```shell
[INFO] [omnigibson] Demo /home/maxliu/projects/OmniGibson-main/omnigibson/examples/scenes/scene_selector.py
    ********************************************************************************
    Description:

    Prompts the user to select any available interactive scene and loads a turtlebot into it.
    It steps the environment 100 times with random actions sampled from the action space,
    using the Gym interface, resetting it 10 times.
    ********************************************************************************

Here is a list of available scene types:

[1] InteractiveTraversableScene: Procedurally generated scene with fully interactive objects

Choose a scene type (enter a number from 1 to 1): 1

Here is a list of available scene models:

[1] Beechwood_0_garden
[2] Beechwood_0_int
[3] Beechwood_1_int
[4] Benevolence_0_int
[5] Benevolence_1_int
[6] Benevolence_2_int
[7] Ihlen_0_int
[8] Ihlen_1_int
[9] Merom_0_garden
[10] Merom_0_int
[11] Merom_1_int
[12] Pomaria_0_garden
[13] Pomaria_0_int
[14] Pomaria_1_int
[15] Pomaria_2_int
[16] Rs_garden
[17] Rs_int
[18] Wainscott_0_garden
[19] Wainscott_0_int
[20] Wainscott_1_int
[21] grocery_store_asian
[22] grocery_store_cafe
[23] grocery_store_convenience
[24] grocery_store_half_stocked
[25] hall_arch_wood
[26] hall_conference_large
[27] hall_glass_ceiling
[28] hall_train_station
[29] hotel_gym_spa
[30] hotel_suite_large
[31] hotel_suite_small
[32] house_double_floor_lower
[33] house_double_floor_upper
[34] house_single_floor
[35] office_bike
[36] office_cubicles_left
[37] office_cubicles_right
[38] office_large
[39] office_vendor_machine
[40] restaurant_asian
[41] restaurant_brunch
[42] restaurant_cafeteria
[43] restaurant_diner
[44] restaurant_hotel
[45] restaurant_urban
[46] school_biology
[47] school_chemistry
[48] school_computer_lab_and_infirmary
[49] school_geography
[50] school_gym

Choose a scene model (enter a number from 1 to 50): 35

Here is a list of available load modes:

[1] Quick: Only load the building assets (i.e.: the floors, walls, ceilings)
[2] Full: Load all interactive objects in the scene

Choose a load mode (enter a number from 1 to 2): 2 
[INFO] [omnigibson.simulator] ----- Starting OmniGibson. This will take 10-30 seconds... -----
Starting kit application with the following args:  ['/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/apps/omnigibson.kit', '--/app/tokens/exe-path=/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/exts', '--ext-folder', '/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/kit/logs/Kit/Isaac-Sim/4.1/kit_20240913_180726.log
2024-09-13 10:07:26 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.101s] Failed to solve some dependencies locally, syncing with extension registry...
[0.125s] [ext: omni.kit.async_engine-0.0.0] startup
[DEBUG] [asyncio] Using selector: EpollSelector
[0.130s] [ext: omni.client-1.1.0] startup
[0.167s] [ext: omni.kit.registry.nucleus-0.0.0] startup
[INFO] [omni.kit.registry.nucleus.registry_manager] 'trusted' is not set for registry 'kit/default' in '/exts/omni.kit.registry.nucleus/registries'
[INFO] [omni.kit.registry.nucleus.registry_manager] 'trusted' is not set for registry 'kit/sdk' in '/exts/omni.kit.registry.nucleus/registries'
[INFO] [omni.kit.registry.nucleus.registry_manager] 'trusted' is not set for registry 'kit/community' in '/exts/omni.kit.registry.nucleus/registries'
2024-09-13 10:07:30 [4,223ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[omnigibson-2023.1.1] dependency: 'omni.kit.streamsdk.plugins' = { version='=2.5.2' } can't be satisfied. Available versions:
	 - [omni.kit.streamsdk.plugins-4.5.1+106.0.0.lx64.r] (/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.kit.streamsdk.plugins-4.5.1+106.0.0.lx64.r)
	 - [omni.kit.streamsdk.plugins-4.4.0+106.0.0.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-4.3.3+106.0.0.lx64.r] (/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.kit.streamsdk.plugins-4.3.3+106.0.0.lx64.r)
	 - [omni.kit.streamsdk.plugins-3.2.1+105.2.lx64.r] (/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.kit.streamsdk.plugins-3.2.1+105.2.lx64.r)
	 - [omni.kit.streamsdk.plugins-2.6.2+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.1+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.0+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.0.3+105.0.lx64.r] (registry)
 Synced registries:
	 - kit/default         : found 424 packages 
	 - kit/sdk             : found 315 packages 
	 - kit/community       : found 362 packages 

2024-09-13 10:07:30 [4,223ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/examples/scenes/scene_selector.py", line 84, in <module>
    main()
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/examples/scenes/scene_selector.py", line 61, in main
    env = og.Environment(configs=cfg)
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/utils/python_utils.py", line 90, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/envs/env_base.py", line 41, in __init__
    launch_simulator()
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/simulator.py", line 173, in launch_simulator
    og.app = _launch_app()
  File "/home/maxliu/projects/OmniGibson-main/omnigibson/simulator.py", line 90, in _launch_app
    app = lazy.omni.isaac.kit.SimulationApp(config_kwargs, experience=str(kit_file_target.resolve(strict=True)))
  File "/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 213, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/maxliu/.local/share/ov/pkg/isaac-sim-4.1.0/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
2024-09-13 10:07:30 [4,227ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

## Environment
* Ubuntu 20.04
* Issac sim 4.1.0

wut19, 2024-09-19:
I've met the same issue. Any solutions?

ge79puv, 2024-09-22:
I also met the exact same issue. Any response from the team? Now in Omniverse Launcher Exchange Apps, there is no more Isaac Sim 2023.1.1 available. How can we fix that?

hang-yin, 2024-09-23:
Hi all, thanks for reaching out! We will cut a new release for the recent isaac sim versions by the end of this week. Meanwhile, you can take a look at https://behavior.stanford.edu/omnigibson-develop/getting_started/installation.html for the installation guide with our development branch. When you clone our repo, please check out our `og-develop` branch instead of the main branch so that isaac 4.0 and 4.1 are supported.