StanfordVL/BEHAVIOR-1K #884: ModuleNotFoundError: No module named 'ompl'

f1ren, 2024-09-15:
On Windows, after installing:
`.\scripts\setup.bat`
And running this example:
`python -m omnigibson.examples.action_primitives.solve_simple_task`
Getting a print that ends as follows:
```
Executing controller
Traceback (most recent call last):
  File "C:\Users\Administrator\anaconda3\envs\omnigibson\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Administrator\anaconda3\envs\omnigibson\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Projects\OmniGibson\omnigibson\examples\action_primitives\solve_simple_task.py", line 73, in <module>
    main()
  File "C:\Projects\OmniGibson\omnigibson\examples\action_primitives\solve_simple_task.py", line 63, in main
    execute_controller(controller.apply_ref(StarterSemanticActionPrimitiveSet.GRASP, grasp_obj), env)
  File "C:\Projects\OmniGibson\omnigibson\examples\action_primitives\solve_simple_task.py", line 14, in execute_controller
    for action in ctrl_gen:
  File "C:\Projects\OmniGibson\omnigibson\action_primitives\starter_semantic_action_primitives.py", line 438, in apply_ref
    yield from ctrl(*args)
  File "C:\Projects\OmniGibson\omnigibson\action_primitives\starter_semantic_action_primitives.py", line 658, in _grasp
    yield from self._navigate_if_needed(obj, pose_on_obj=grasp_pose)
  File "C:\Projects\OmniGibson\omnigibson\action_primitives\starter_semantic_action_primitives.py", line 1494, in _navigate_if_needed
    yield from self._navigate_to_obj(obj, pose_on_obj=pose_on_obj, **kwargs)
  File "C:\Projects\OmniGibson\omnigibson\action_primitives\starter_semantic_action_primitives.py", line 1508, in _navigate_to_obj
    yield from self._navigate_to_pose(pose)
  File "C:\Projects\OmniGibson\omnigibson\action_primitives\starter_semantic_action_primitives.py", line 1432, in _navigate_to_pose
    plan = plan_base_motion(
  File "C:\Projects\OmniGibson\omnigibson\utils\motion_planning_utils.py", line 47, in plan_base_motion
    from ompl import base as ob
ModuleNotFoundError: No module named 'ompl'
```

Tried to install using `conda install conda-forge::ompl` which breaks the environment, making the OmniGibson fail fast on imports.
Also tried using `conda install --no-deps conda-forge::ompl` which did not solve the original problem.

cgokmen, 2024-09-15:
Hi, the action primitives use the OMPL library. At the time of development , this library was relatively difficult to install on an arbitrary non-system Python environment due to system-level dependencies. As a result, we recommend trying to run the example in our Docker container which includes OMPL installed. Otherwise you can go to the OMPL repository for more info for how to install it locally.

closed 2024-09-15 by cgokmen

yanan1116, 2025-04-04:
any updates on how to install it without using docker ?