StanfordVL/BEHAVIOR-1K #892: Is it normal to be unable to complete the navigation for a long time?

RookieXwc, 2024-09-19:
[image]
@cgokmen I just got in touch with omniibson and ran `navigation_devv_demo`. But I found that when the initial position, target position, and trajectory have been well visualized, the turtlebox seems to spin in place instead of quickly reaching the target position according to the trajectory. Is this normal? Perhaps 100000 timesteps have already been run. My log is as follows:
[WARNING] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
2024-09-19 03:25:47 [37,490ms] [Warning] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
[INFO] [omnigibson.simulator] Imported scene.
[INFO] [omnigibson.objects.object_base] Loaded robot0 at /World/robot0
[INFO] [omnigibson.objects.controllable_object] Control frequency is None - being set to default of render_frequency: 60.0000
2024-09-19 03:25:51 [41,639ms] [Warning] [omni.syntheticdata.plugin] SdRenderVarPtr missing valid input renderVar DistanceToCameraSDhost
[WARNING] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
2024-09-19 03:25:52 [41,913ms] [Warning] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
[WARNING] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:52 [41,934ms] [Warning] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.
[INFO] [omnigibson.objects.object_base] Loaded task_initial_pos_marker at /World/task_initial_pos_marker
[INFO] [omnigibson.objects.object_base] Loaded task_goal_pos_marker at /World/task_goal_pos_marker
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker0 at /World/task_waypoint_marker0
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker1 at /World/task_waypoint_marker1
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker2 at /World/task_waypoint_marker2
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker3 at /World/task_waypoint_marker3
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker4 at /World/task_waypoint_marker4
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker5 at /World/task_waypoint_marker5
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker6 at /World/task_waypoint_marker6
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker7 at /World/task_waypoint_marker7
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker8 at /World/task_waypoint_marker8
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker9 at /World/task_waypoint_marker9
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker10 at /World/task_waypoint_marker10
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker11 at /World/task_waypoint_marker11
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker12 at /World/task_waypoint_marker12
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker13 at /World/task_waypoint_marker13
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker14 at /World/task_waypoint_marker14
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker15 at /World/task_waypoint_marker15
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker16 at /World/task_waypoint_marker16
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker17 at /World/task_waypoint_marker17
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker18 at /World/task_waypoint_marker18
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker19 at /World/task_waypoint_marker19
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker20 at /World/task_waypoint_marker20
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker21 at /World/task_waypoint_marker21
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker22 at /World/task_waypoint_marker22
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker23 at /World/task_waypoint_marker23
[INFO] [omnigibson.objects.object_base] Loaded task_waypoint_marker24 at /World/task_waypoint_marker24
2024-09-19 03:25:56 [46,016ms] [Warning] [carb] Client omni.physx.tensors.plugin has acquired [omni::physx::IPhysxSimulation v1.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
[WARNING] [omnigibson.utils.registry_utils] Object 'task_initial_pos_marker' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,064ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_initial_pos_marker' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_goal_pos_marker' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_goal_pos_marker' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker0' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker0' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker1' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker1' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker2' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker2' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker3' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker3' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker4' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker4' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker5' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker5' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker6' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker6' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker7' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker7' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker8' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker8' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker9' is not in the state dict to load from. Skip loading its state.
2024-09-19 03:25:56 [46,065ms] [Warning] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker9' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.utils.registry_utils] Object 'task_waypoint_marker10' is not in the state dict to load from. Skip loading its state.
[... cut ...]
[INFO] [omnigibson.tasks.point_navigation_task] Sampled initial pose: [-2.6  1.9  0. ], [ 0.          0.          0.4361277  -0.89988479]
[INFO] [omnigibson.tasks.point_navigation_task] Sampled goal position: [-2.9  2.8  0. ]
[INFO] [omnigibson.tasks.point_navigation_task] Sampled initial pose: [1.1 2.4 0. ], [ 0.          0.          0.5118192  -0.85909319]
[INFO] [omnigibson.tasks.point_navigation_task] Sampled goal position: [-1.1  2.3  0. ]
[INFO] [omnigibson.utils.sim_utils] Landed object robot0 successfully!

RookieXwc, 2024-09-20:
When I run it under `Isaac Sim version 4.1` and the `og-develop` branch, the results are similar. The turtlebot is almost spinning in place instead of navigating along the path.

ChengshuLi, 2024-09-23:
Thanks for the question! This behavior is expected. If you look at the code, the robot is simply outputting random actions (therefore, it appears to be spinning around in place). For actual navigation behavior of following waypoints, you can look at our action primitives.

closed 2024-09-23 by ChengshuLi