StanfordVL/BEHAVIOR-1K #893: TypeError: XFormPrimView.get_world_poses() got an unexpected keyword argument 'usd'

RookieXwc, 2024-09-19:
When I run examples `python -m omnigibson.examples.robots.robot_control_example` on the `og-develop` branch, the following error occurs:
```
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/objects/stateful_object.py", line 144, in _post_load
    super()._post_load()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/objects/object_base.py", line 161, in _post_load
    super()._post_load()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/prims/entity_prim.py", line 127, in _post_load
    self.update_links()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/prims/entity_prim.py", line 257, in update_links
    self._links[link_name].load(self.scene)
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/prims/prim_base.py", line 117, in load
    self._post_load()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/prims/rigid_prim.py", line 93, in _post_load
    self._rigid_prim_view_direct = RigidPrimView(self.prim_path)
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/rigid_prim_view.py", line 144, in __init__
    XFormPrimView.__init__(
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/xform_prim_view.py", line 154, in __init__
    self._set_xform_properties()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/prims/xform_prim_view.py", line 262, in _set_xform_properties
    current_positions, current_orientations = self.get_world_poses()
  File "/home/gpu/.local/share/ov/pkg/isaac-sim-2023.1.1/OmniGibson/omnigibson/utils/deprecated_utils.py", line 883, in get_world_poses
    return _XFormPrimView.get_world_poses(self, indices=indices, usd=usd)
TypeError: XFormPrimView.get_world_poses() got an unexpected keyword argument 'usd'
```
@cgokmen  This indicates that the `usd` parameter was used when calling `_XFormPrimView.get_corld_poses` in `OmniGibson/mnigibson/tiles/deptrecated_utils_py`. My isaac sim version is 2023.1.1, and the `get-world_poses` function in `isaac-sim-2023.1.1/extts/omni.isaac-core/omni/isaac/core/prems/xform_prim-view.py` does not have a usd parameter:
[image]

cgokmen, 2024-09-19:
Hi, on og-develop we are now tracking the latest Isaac Sim version which is 4.1.0. Please try on that version.

RookieXwc, 2024-09-19:
@cgokmen  Thank you for your quick reply! I encountered the same issue as #805 under the `main` branch, regarding `examples.objects.draw_bounding_box`. Following the plan  #805, I switched to the `og-develop` branch. Are you saying that if I want to run `examples.objects.draw_bounding_box`, I need to switch to the `og-develop` branch and use Isaac Sim 4.1.0?

cgokmen, 2024-09-19:
Yes, sorry for this - the long story short is that the example was broken in main, so it was fixed in og-develop for our next release, but also on og-develop (separately from this issue) we have upgraded to 4.1. So if you wish to use og-develop you need to be on Isaac Sim 4+.

closed 2024-09-19 by cgokmen