StanfordVL/BEHAVIOR-1K #898: Installation of Isaac Sim 2023.1.1 through Omniverse Launcher Exchange Apps

ge79puv, 2024-09-22:
Hey there,

I am trying to install Isaac Sim 2023.1.1 through Omniverse Launcher Exchange Apps. However, there is no this version on Omniverse Launcher any more. Could you tell me how to install this version of Isaac Sim?
[image]

Thanks a lot in advance!

BR

hang-yin, 2024-09-23:
Hi @ge79puv , thanks for reaching out! We will cut a new release for the recent isaac sim versions by the end of this week. Meanwhile, you can take a look at https://behavior.stanford.edu/omnigibson-develop/getting_started/installation.html for the installation guide with our development branch. When you clone our repo, please check out our `og-develop` branch instead of the main branch so that isaac 4.0 and 4.1 are supported.

closed 2024-09-24 by ge79puv

mengxyokok, 2024-09-30:
isaac 4.2  is surpported by isaac_4_2 branch  ??