StanfordVL/BEHAVIOR-1K #899: how can I install OmniGibson if I installed issac-sim through pip?

DawnyWu, 2024-09-22:
I install issac-sim 4.0 by using pip like this https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html 

so I don't have a folder like `~/.local/share/ov/pkg/Isaac-sim` and no `isaac*.sh` file, how can i install omnigibson? 

and when will the new version of omnigibson release because we cannot install isaac-sim 2023.1.1 anymore. 

Thank you.

hang-yin, 2024-09-23:
Hi @DawnyWu , thanks for reaching out! We will cut a new release for the recent isaac sim versions by the end of this week. Meanwhile, you can take a look at https://behavior.stanford.edu/omnigibson-develop/getting_started/installation.html for the installation guide with our development branch. When you clone our repo, please check out our `og-develop` branch instead of the main branch so that isaac 4.0 is supported.

DawnyWu, 2024-09-24:
Thank you so much

closed 2024-09-24 by DawnyWu