StanfordVL/BEHAVIOR-1K #914: AttributeError: module lazy_ has no attribute omni

JackieXuu, 2024-09-29:
Hi, I have installed this, but when I run example, it raises error
```
  File "/home//code/OmniGibson/omnigibson/utils/lazy_import_utils.py", line 29, in __getattr__
    raise AttributeError(
AttributeError: module lazy_ has no attribute omni

```

Do you guys know how to fix it?

cgokmen, 2024-09-29:
Hi, this is not the actual error, it's the final symptom of Isaac Sim not being discoverable / launching properly. Please send the full output.

JackieXuu, 2024-09-29:
> Hi, this is not the actual error, it's the final symptom of Isaac Sim not being discoverable / launching properly. Please send the full output.

```
Here is a list of available robots:

[1] BehaviorRobot
[2] Fetch
[3] FrankaAllegro
[4] FrankaLeap
[5] FrankaPanda
[6] Freight
[7] Husky
[8] Locobot
[9] Tiago
[10] Turtlebot
[11] VX300S

Choose a robot (enter a number from 1 to 11): 4  
[INFO] [omnigibson.simulator] ----- Starting OmniGibson. This will take 10-30 seconds... -----
Traceback (most recent call last):
  File "/home/ask/anaconda3/envs/omnigibson/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/ask/anaconda3/envs/omnigibson/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 154, in <module>
    main()
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 93, in main
    env = og.Environment(configs=cfg)
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/utils/python_utils.py", line 90, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/envs/env_base.py", line 41, in __init__
    launch_simulator()
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/simulator.py", line 173, in launch_simulator
    og.app = _launch_app()
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/simulator.py", line 89, in _launch_app
    with launch_context(None):
  File "/home/ask/anaconda3/envs/omnigibson/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/utils/ui_utils.py", line 165, in suppress_omni_log
    log = lazy.omni.log.get_log()
  File "/home/ask/askTeam/code/OmniGibson/omnigibson/utils/lazy_import_utils.py", line 29, in __getattr__
    raise AttributeError(
AttributeError: module lazy_ has no attribute omni

```
we need Isaac sim 2023.1.1 right?

hang-yin, 2024-10-02:
@JackieXuu OmniGibson v1.1.0 was released yesterday, please check [here](https://github.com/StanfordVL/OmniGibson/releases/tag/v1.1.0) for our release notes and [here](https://behavior.stanford.edu/omnigibson/getting_started/installation.html) for our updated documentation on how to install (& more!) - note that this release is compatible with Isaac Sim 4.0.0 and 4.1.0 but not 4.2.0 yet. Let us know here if this fixes your problem!