StanfordVL/BEHAVIOR-1K #940: Error when running python -m omnigibson.examples.learning.navigation_policy_demo

ge79puv, 2024-10-08:
Hey there, 

I just used Isaac Sim 4.1.0 and the new released OmniGibson v 1.1.1. I tried to run the demo of RL. However, there is an error: 

[image]

Could you help me to solve this issue?

Best regards

ChengshuLi, 2024-10-08:
Could you let us know the version of `gymnasium`?

ge79puv, 2024-10-08:
0.29.1

ChengshuLi, 2024-10-08:
Hmm I am actually not able to reproduce your issue. Maybe you can take a closer look at `obs.spaces`? If it's not `OrderedDict`, what is it?

ge79puv, 2024-10-09:
I downgrade my gymnasium version, now it works. Thanks a lot!

closed 2024-10-09 by ge79puv

ge79puv, 2024-10-09:
Hi,

this omnigibson.examples.learning.navigation_policy_demo sometimes runs fine, but sometimes it suddenly terminates after running for a while. I have not modified any source code, and the error message is as follows: 
[image]
I cannot understand why the tensors are loaded on two gpus. And this termination is uncertain. Sometimes this error appears at the beginning, and sometimes it appears after running for a period of time, which makes me unable to complete the entire RL training.

Best regards