StanfordVL/BEHAVIOR-1K #973: How to use custom .usd files？

johnbhlm, 2024-10-14:
Hi,
thanks for great work!

 We are following the work of ReKep, which was built based on the work omnigibson, and we would like to work on building our own simulation environment here as well.

Using the Robot Grasping Demo example（grasping_mode_example.py）, we added our object to the scene, and when we loaded our custom .usd file there was a problem, as shown in the code and the video.

```
"""
Example script demo'ing robot manipulation control with grasping.
"""
import torch as th

import omnigibson as og
from omnigibson.macros import gm
from omnigibson.sensors import VisionSensor
from omnigibson.utils.ui_utils import KeyboardRobotController, choose_from_options

GRASPING_MODES = dict(
    sticky="Sticky Mitten - Objects are magnetized when they touch the fingers and a CLOSE command is given",
    assisted="Assisted Grasping - Objects are magnetized when they touch the fingers, are within the hand, and a CLOSE command is given",
    physical="Physical Grasping - No additional grasping assistance applied",
)

# Don't use GPU dynamics and Use flatcache for performance boost
gm.USE_GPU_DYNAMICS = False
gm.ENABLE_FLATCACHE = True

def main(random_selection=False, headless=False, short_exec=False):
    """
    Robot grasping mode demo with selection
    Queries the user to select a type of grasping mode
    """
    og.log.info(f"Demo {__file__}\n    " + "*" * 80 + "\n    Description:\n" + main.__doc__ + "*" * 80)

    # Choose type of grasping
    grasping_mode = choose_from_options(options=GRASPING_MODES, name="grasping mode", random_selection=random_selection)

    # Create environment configuration to use
    scene_cfg = dict(type="Scene")
    robot0_cfg = dict(
        type="Fetch",
        obs_modalities=["rgb"],  # we're just doing a grasping demo so we don't need all observation modalities
        action_type="continuous",
        action_normalize=True,
        grasping_mode=grasping_mode,
    )

    # Define objects to load
    table_cfg = dict(
        type="DatasetObject",
        name="table",
        category="breakfast_table",
        model="lcsizg",
        bounding_box=[0.5, 0.5, 0.8],
        fixed_base=True,
        position=[0.7, -0.1, 0.6],
        orientation=[0, 0, 0.707, 0.707],
    )

    chair_cfg = dict(
        type="DatasetObject",
        name="chair",
        category="straight_chair",
        model="amgwaw",
        bounding_box=None,
        fixed_base=False,
        position=[0.45, 0.65, 0.425],
        orientation=[0, 0, -0.9990215, -0.0442276],
    )

    box_cfg = dict(
        type="PrimitiveObject",
        name="box",
        primitive_type="Cube",
        rgba=[1.0, 0, 0, 1.0],
        size=0.05,
        position=[0.53, -0.1, 0.97],
    )
    # our object
    leg1_cfg = dict(
        type = "USDObject",
        name= "square_table_leg1",
        usd_path= f"/home/furniture-bench/furniture_bench/assets/furniture/mesh/square_table/square_table_leg1.usd",
        scale= 2.0,
        position = [0.59, -0.1, 0.97],
    )

    # Compile config
    cfg = dict(scene=scene_cfg, robots=[robot0_cfg], objects=[table_cfg, chair_cfg, box_cfg,leg1_cfg])

    # Create the environment
    env = og.Environment(configs=cfg)

    # Reset the robot
    robot = env.robots[0]
    robot.set_position_orientation(position=[0, 0, 0])
    robot.reset()
    robot.keep_still()

    # Make the robot's camera(s) high-res
    for sensor in robot.sensors.values():
        if isinstance(sensor, VisionSensor):
            sensor.image_height = 720
            sensor.image_width = 720

    # Update the simulator's viewer camera's pose so it points towards the robot
    og.sim.viewer_camera.set_position_orientation(
        position=th.tensor([-2.39951, 2.26469, 2.66227]),
        orientation=th.tensor([-0.23898481, 0.48475231, 0.75464013, -0.37204802]),
    )

    # Create teleop controller
    action_generator = KeyboardRobotController(robot=robot)

    # Print out relevant keyboard info if using keyboard teleop
    action_generator.print_keyboard_teleop_info()

    # Other helpful user info
    print("Running demo with grasping mode {}.".format(grasping_mode))
    print("Press ESC to quit")

    # Loop control until user quits
    max_steps = -1 if not short_exec else 100
    step = 0
    while step != max_steps:
        action = action_generator.get_random_action() if random_selection else action_generator.get_teleop_action()
        for _ in range(10):
            env.step(action)
            step += 1

    # Always shut down the environment cleanly at the end
    og.clear()

if __name__ == "__main__":
    main()
```

This is the code where we add the object:
```
 # our object
    leg1_cfg = dict(
        type = "USDObject",
        name= "square_table_leg1",
        usd_path= f"/home/furniture-bench/furniture_bench/assets/furniture/mesh/square_table/square_table_leg1.usd",
        scale= 2.0,
        position = [0.59, -0.1, 0.97],
    )
```

The results of the run are as follows:

[image]

[Screencast from 2024年10月14日 15时38分28秒.webm](https://github.com/user-attachments/assets/2ed4dd08-9ec0-421c-82a9-820150e0355b)

We found that our object passes through the table and does not produce realistic physics (collision like the red cube), so we would like to know how to load our .usd file that can produce the same collision effect as the red cube?

The .usd file provided by our open source repository：[object  .usd file link](https://github.com/clvrai/furniture-bench/tree/main/furniture_bench/assets/furniture/mesh/square_table),  

Looking forward to your reply~

ChengshuLi, 2024-10-14:
You shoud start a vanilla Isaac Sim and open your custom USD file.
Then you will need to add Rigid Body API to each link of the object, and add Collision Body API to each collision mesh of each link.

closed 2024-10-14 by ChengshuLi

johnbhlm, 2024-10-15:
Thank you for your reply.

When we add Rigid Body API to the object in a vanilla isaac sim, the steps as below:
1 open one_leg.usd file
2. select  the object
3. add -->Physics -->Rigid Body with Colliders Preset
4. save the .usd file
as show the video
[2024年10月15日 屏幕视频 19时07分52秒.webm](https://github.com/user-attachments/assets/93d2f8fc-7967-44eb-9254-2efd40d31308)

Then we can generate real collision effect in isaac sim, as show below:
[2024年10月15日 屏幕视频 19时19分54秒.webm](https://github.com/user-attachments/assets/cdf61afb-7ae4-4378-9074-b0604acc35f3)

Then we load it  in omnigibson, the object becomes transparent, but the effect can be seen in the UI by displaying the physical collision, as shown in the video below:
[Screencast from 2024年10月14日 15时38分28秒.webm](https://github.com/user-attachments/assets/e1ec6e57-9be6-4fea-8fda-6beff0e2abfd)

[image]

Now I don't know if there is a problem adding the rigid body api in isaac sim or a problem loading the .usd in omnigibson or if there is a problem with the original .usd file that ultimately leads to incorrect results.

I hope to get your reply and guidance, thanks again~

johnbhlm, 2024-10-15:
You can verify the result if you can, the original .usd file is here: [https://github.com/clvrai/furniture-bench/tree/main/furniture_bench/assets/furniture/mesh/square_table](https://github.com/clvrai/furniture-bench/tree/main/furniture_bench/assets/furniture/mesh/square_table)

johnbhlm, 2024-10-16:
Hello.
I found that in the .usd file saved after adding the rigid collision api to an object in isaac sim, the visual ->Purposes for each mesh is default; as show the bellow image:
[image]

But when loaded in omnigibson, this property changes to guide, so that only the collision lines are visible, not the meshes. as show the bellow image:
[image]

So, did omnigibson do something to the .usd file that caused the property to change?

What should I do to keep the original properties?

Hope to get your reply and answer, thanks again.