StanfordVL/BEHAVIOR-1K #974: Question on Segmentation fault after FileNotFoundError: Ihlen_0_int_task_store_firewood_0_0_template.json

Roythescientist, 2024-10-14:
Hi, every Omnigibson specialist. Currently, I'm testing the [DK-Prompt](https://github.com/keke-220/vlm-tamp) project. After I successfully installed the library that is required to run the program. I use this command python eval.py. The error indicated that the render is not the multiplication of physics dt. I set the physics frequency from 100 to 60 in the config. After that, a new error emerge as a Segmentation fault (core dumped). I discovered a FFileNotFoundError: [Errno 2] No such file or directory: '/home/Austine/OmniGibson/omnigibson/data/og_dataset/scenes/Ihlen_0_int/json/Ihlen_0_int_task_store_firewood_0_0_template.json'. Can anyone show me if there is a way I can solve this because I have even installed BEHAVIOR-1K dataset by running python omnigibson/download_datasets.py but I still get the same error

Roythescientist, 2024-10-14:
This is the log:
Traceback (most recent call last):
  File "/home/Austine/OmniGibson/vlm-tamp/eval.py", line 1060, in <module>
    env = og.Environment(configs=config)
  File "/home/Austine/OmniGibson/omnigibson/utils/python_utils.py", line 93, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/Austine/OmniGibson/omnigibson/envs/env_base.py", line 128, in __init__
    self.load()
  File "/home/Austine/OmniGibson/omnigibson/envs/env_base.py", line 429, in load
    self._load_scene()
  File "/home/Austine/OmniGibson/omnigibson/envs/env_base.py", line 251, in _load_scene
    self._scene = create_class_from_registry_and_config(
  File "/home/Austine/OmniGibson/omnigibson/utils/python_utils.py", line 286, in create_class_from_registry_and_config
    return cls(**cls_kwargs)
  File "/home/Austine/OmniGibson/omnigibson/utils/python_utils.py", line 93, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/Austine/OmniGibson/omnigibson/scenes/interactive_traversable_scene.py", line 93, in __init__
    super().__init__(
  File "/home/Austine/OmniGibson/omnigibson/utils/python_utils.py", line 71, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/Austine/OmniGibson/omnigibson/scenes/traversable_scene.py", line 50, in __init__
    super().__init__(
  File "/home/Austine/OmniGibson/omnigibson/utils/python_utils.py", line 71, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/Austine/OmniGibson/omnigibson/scenes/scene_base.py", line 112, in __init__
    with open(self.scene_file, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/Austine/OmniGibson/omnigibson/data/og_dataset/scenes/Ihlen_0_int/json/Ihlen_0_int_task_store_firewood_0_0_template.json'
2024-10-14 16:34:20 [6,764ms] [Warning] [carb] [Plugin: omni.sensors.tiled.plugin] Module /home/Austine/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/extscache/omni.sensors.tiled-0.0.4+106.0.0.lx64.r/bin/libomni.sensors.tiled.plugin.so remained loaded after unload request
2024-10-14 16:34:20 [6,769ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/Austine/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/extscache/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,771ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,772ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,772ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-14 16:34:20 [6,772ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-10-14 16:34:21 [6,896ms] [Warning] [carb] Client omni.syntheticdata.plugin Failed to acquire interface [omni::graph::core::INode v4.10] while unloading all plugins
Fatal Python error: Segmentation fault

Current thread 0x00007fee52e00180 (most recent call first):
  <no Python frame>

Extension modules: yaml._yaml, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, h5py._errors, h5py.defs, h5py._objects, h5py.h5, h5py.utils, h5py.h5t, h5py.h5s, h5py.h5ac, h5py.h5p, h5py.h5r, h5py._proxy, h5py._conv, h5py.h5z, h5py.h5a, h5py.h5d, h5py.h5ds, h5py.h5g, h5py.h5i, h5py.h5f, h5py.h5fd, h5py.h5pl, h5py.h5o, h5py.h5l, h5py._selector, PIL._imaging, scipy._lib._ccallback_c, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest [... cut ...] ats._sobol, scipy.stats._qmc_cy, scipy.stats._rcont.rcont, kiwisolver._cext, PIL._imagingft, gmpy2.gmpy2, _brotli, grpc._cython.cygrpc, google._upb._message, quaternion.numpy_quaternion, omni.mdl.pymdlsdk._pymdlsdk, osqp._osqp, multidict._multidict, yarl._quoting_c, aiohttp._helpers, aiohttp._http_writer, aiohttp._http_parser, aiohttp._websocket, frozenlist._frozenlist, scipy.io.matlab._mio_utils, scipy.io.matlab._streams, scipy.io.matlab._mio5_utils (total: 172)
Segmentation fault (core dumped)

ChengshuLi, 2024-10-14:
For each task (such as store_firewood), we currently only provide one pre-sampled initial configuration. Specifically for store_firewood, we have `Merom_0_int/json/Merom_0_int_task_store_firewood_0_0_template.json`.

However, you can still online sample a task in any scene of your choice, with `online_sampling=True` in `BehaviorTask` and specifying the scene name to `Ihlen_0_int`, for instance.

closed 2024-10-14 by ChengshuLi