StanfordVL/BEHAVIOR-1K #976: Add script for converting plain USDs into OmniGibson object format

cgokmen, 2024-10-14:


cgokmen, 2024-10-21:
This would help things like #991.

johnbhlm, 2024-10-30:
Hello,
    Do you have provided  a script that rearranges an arbitrary object into the appropriate hierarchy?
For example, usd in [https://github.com/clvrai/furniture-bench/tree/main/furniture_bench/assets/furniture/mesh/square_table](https://github.com/clvrai/furniture-bench/tree/main/furniture_bench/assets/furniture/mesh/square_tablel)
    I have tried using the method in #991  but still no success.
    
   I hope to get your support and help. Thanks again.

albertcity, 2024-11-26:
@johnbhlm @cgokmen Any updates about the issue?

cgokmen, 2024-11-26:
We are still actively working on this. Currently no ETA but we should be able to ship it this year. Will keep you updated.

cgokmen, 2024-11-26:
@cremebrule fyi

jinkun-hao, 2024-12-23:
This script would be very helpful. Is there any progress? @cgokmen

cremebrule, 2024-12-24:
Hi @jinkun-hao ,

Apologies for the delay! We've integrated this feature into our `og-develop` branch, and is now avaiable as the [`import_custom_object.py` example script](https://github.com/StanfordVL/OmniGibson/blob/og-develop/omnigibson/examples/objects/import_custom_object.py)!