StanfordVL/BEHAVIOR-1K #991: Load custom USD file

johnbhlm, 2024-10-18:
Hi,
thanks for great work!

We are following the work of ReKep, which was built based on the work omnigibson, and we would like to work on building our own simulation environment here as well.

Using the Robot Grasping Demo example（grasping_mode_example.py）, we added our object to the scene, as show bellow:
```
"""
Example script demo'ing robot manipulation control with grasping.
"""
import torch as th

import omnigibson as og
from omnigibson.macros import gm
from omnigibson.sensors import VisionSensor
from omnigibson.utils.ui_utils import KeyboardRobotController, choose_from_options

GRASPING_MODES = dict(
    sticky="Sticky Mitten - Objects are magnetized when they touch the fingers and a CLOSE command is given",
    assisted="Assisted Grasping - Objects are magnetized when they touch the fingers, are within the hand, and a CLOSE command is given",
    physical="Physical Grasping - No additional grasping assistance applied",
)

# Don't use GPU dynamics and Use flatcache for performance boost
gm.USE_GPU_DYNAMICS = False
gm.ENABLE_FLATCACHE = True

def main(random_selection=False, headless=False, short_exec=False):
    """
    Robot grasping mode demo with selection
    Queries the user to select a type of grasping mode
    """
    og.log.info(f"Demo {__file__}\n    " + "*" * 80 + "\n    Description:\n" + main.__doc__ + "*" * 80)

    # Choose type of grasping
    grasping_mode = choose_from_options(options=GRASPING_MODES, name="grasping mode", random_selection=random_selection)

    # Create environment configuration to use
    scene_cfg = dict(type="Scene")
    robot0_cfg = dict(
        type="Fetch",
        obs_modalities=["rgb"],  # we're just doing a grasping demo so we don't need all observation modalities
        action_type="continuous",
        action_normalize=True,
        grasping_mode=grasping_mode,
    )

    # Define objects to load
    table_cfg = dict(
        type="DatasetObject",
        name="table",
        category="breakfast_table",
        model="lcsizg",
        bounding_box=[0.5, 0.5, 0.8],
        fixed_base=True,
        position=[0.7, -0.1, 0.6],
        orientation=[0, 0, 0.707, 0.707],
    )

    chair_cfg = dict(
        type="DatasetObject",
        name="chair",
        category="straight_chair",
        model="amgwaw",
        bounding_box=None,
        fixed_base=False,
        position=[0.45, 0.65, 0.425],
        orientation=[0, 0, -0.9990215, -0.0442276],
    )

    box_cfg = dict(
        type="PrimitiveObject",
        name="box",
        primitive_type="Cube",
        rgba=[1.0, 0, 0, 1.0],
        size=0.05,
        position=[0.53, -0.1, 0.97],
    )
    # our object
    leg1_cfg = dict(
        type = "USDObject",
        name= "square_table_leg1",
        usd_path= f"/home/furniture-bench/furniture_bench/assets/furniture/mesh/square_table/square_table_leg1.usd",
        scale= 2.0,
        position = [0.59, -0.1, 0.97],
    )

    # Compile config
    cfg = dict(scene=scene_cfg, robots=[robot0_cfg], objects=[table_cfg, chair_cfg, box_cfg,leg1_cfg])

    # Create the environment
    env = og.Environment(configs=cfg)

    # Reset the robot
    robot = env.robots[0]
    robot.set_position_orientation(position=[0, 0, 0])
    robot.reset()
    robot.keep_still()

    # Make the robot's camera(s) high-res
    for sensor in robot.sensors.values():
        if isinstance(sensor, VisionSensor):
            sensor.image_height = 720
            sensor.image_width = 720

    # Update the simulator's viewer camera's pose so it points towards the robot
    og.sim.viewer_camera.set_position_orientation(
        position=th.tensor([-2.39951, 2.26469, 2.66227]),
        orientation=th.tensor([-0.23898481, 0.48475231, 0.75464013, -0.37204802]),
    )

    # Create teleop controller
    action_generator = KeyboardRobotController(robot=robot)

    # Print out relevant keyboard info if using keyboard teleop
    action_generator.print_keyboard_teleop_info()

    # Other helpful user info
    print("Running demo with grasping mode {}.".format(grasping_mode))
    print("Press ESC to quit")

    # Loop control until user quits
    max_steps = -1 if not short_exec else 100
    step = 0
    while step != max_steps:
        action = action_generator.get_random_action() if random_selection else action_generator.get_teleop_action()
        for _ in range(10):
            env.step(action)
            step += 1

    # Always shut down the environment cleanly at the end
    og.clear()

if __name__ == "__main__":
    main()
```
This is the code where we add the object:
```
 # our object
    leg1_cfg = dict(
        type = "USDObject",
        name= "square_table_leg1",
        usd_path= f"/home/furniture-bench/furniture_bench/assets/furniture/mesh/square_table/square_table_leg1.usd",
        scale= 2.0,
        position = [0.59, -0.1, 0.97],
    )
```

Then we load it in omnigibson, the object becomes transparent, but the effect can be seen in the UI by displaying the physical collision, as shown in the video below:
[Uploading 2024年10月15日 屏幕视频 14时45分20秒.webm…]()

When I check the property, find that has changed to guide, so that only the collision lines are visible, not the meshes. as show the bellow image:
[image]

So, did omnigibson do something to the .usd file that caused the property to change?

What should I do to keep the original properties?

**Desktop (please complete the following information):**
 - OS:  Ubuntu 22.04
 - Isaac Sim Version 2023.
 - OmniGibson Version 1.0.0

Hope to get your reply and answer, thanks again.

cgokmen, 2024-10-21:
The long story short here is that OmniGibson expects a particular hierarchy for its objects: you have an object prim (which is an xform prim and the default prim in the object USD file), under that you have an xform prim for each link (these are the prims that have the Rigid Body API attached to them), and under that prim you have a "visuals" prim that contains all the visual meshes, and a "collisions" prim that contains all the collision meshes (the stuff with the Collision API). Our code assumes that all prims are EITHER collision OR visual and NOT both like in your case. So this prim gets recognized as a collision prim and then gets hidden by setting the purpose. You can solve this by duplicating the prim, attaching one copy with the Collision API and not the other, and effectively they will be the collision and visual prims. 

I will look into this later this week to provide users with a script that rearranges an arbitrary object into the appropriate hierarchy.

closed 2024-10-21 by cgokmen

johnbhlm, 2024-10-24:
> The long story short here is that OmniGibson expects a particular hierarchy for its objects: you have an object prim (which is an xform prim and the default prim in the object USD file), under that you have an xform prim for each link (these are the prims that have the Rigid Body API attached to them), and under that prim you have a "visuals" prim that contains all the visual meshes, and a "collisions" prim that contains all the collision meshes (the stuff with the Collision API). Our code assumes that all meshes are EITHER collision OR visual and NOT both like in your case. So this prim gets recognized as a collision prim and then gets hidden by setting the purpose. You can solve this by duplicating the prim, attaching one copy with the Collision API and not the other, and effectively they will be the collision and visual prims.
> 
> I will look into this later this week to provide users with a script that rearranges an arbitrary object into the appropriate hierarchy.

Thank you very much for your reply and explanation.
At the same time, I am looking forward to your public script so that everyone can freely convert custom USD and use it in OmniGibson.