StanfordVL/BEHAVIOR-1K #1048: A cuda toolkit version setting might be changed and cause "unable to get a valid CUDA device id from the renderer. Defaulting to GPU0."

Snowflake-Pink, 2024-12-06:
**Describe the bug**
A cuda toolkit version setting might changed while verifing installation by `python -m omnigibson.examples.robots.robot_control_example --quickstart`.

[image]

And it may have caused the subsequent infinite loop later: 
```
2024-12-06 08:41:50 [8,581ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-12-06 08:41:50 [8,582ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
2024-12-06 08:41:50 [8,583ms] [Warning] [omni.graph.core.plugin] OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.
````

**To Reproduce**
Steps to reproduce the behavior:
1. Run `python -m omnigibson.examples.robots.robot_control_example --quickstart`

My environment variable: (**`OMNIGIBSON_HEADLESS=1`**)

```
export OMNIGIBSON_ASSET_PATH="/root/autodl-tmp/digital-cousins/omnigibson_data/assets"
export OMNIGIBSON_DATASET_PATH="/root/autodl-tmp/digital-cousins/omnigibson_data/og_dataset"
export HF_ENDPOINT=https://hf-mirror.com
export SAM2_REPO_ROOT=/root/autodl-tmp/digital-cousins/deps/segment-anything-2/sam2
export PYTHONPATH="${SAM2_REPO_ROOT}:${PYTHONPATH}"
export OMNI_KIT_ALLOW_ROOT=1
export OMNI_KIT_ACCEPT_EULA=YES
export OMNIGIBSON_HEADLESS=1
export XDG_CACHE_HOME=/root/autodl-tmp/.cache
export LD_LIBRARY_PATH=/root/miniconda3/envs/acdc/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin:$LD_LIBRARY_PATH
export OMNIGIBSON_GPU_ID=0
```

My cuda environment:

```
(acdc) root@autodl-container-07904e86c4-e8dc2739:~# nvidia-smi
Fri Dec  6 17:04:26 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090        On  |   00000000:1F:00.0 Off |                  Off |
| 30%   29C    P8             20W /  450W |       1MiB /  24564MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
(acdc) root@autodl-container-07904e86c4-e8dc2739:~# nvcc -V   
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Mar_28_02:18:24_PDT_2024
Cuda compilation tools, release 12.4, V12.4.131
Build cuda_12.4.r12.4/compiler.34097967_0
```

**Expected behavior**
Don't change the toolkit version.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 22.04]
 - Isaac Sim Version [e.g. 4.1.0]
 - OmniGibson Version [e.g. 1.1.1]

**Additional context**
Some other errors but might be caused by no moniter display: (I had set **`OMNIGIBSON_HEADLESS=1`**)

[image]

**Full logs**

Snowflake-Pink, 2024-12-06:
**Logs**

[logs.txt](https://github.com/user-attachments/files/18035536/logs.txt)

Snowflake-Pink, 2024-12-06:
[image]

`APP Ready`↑

Snowflake-Pink, 2024-12-07:
[image]

[cremebrule/digital-cousins/issues/16](https://github.com/cremebrule/digital-cousins/issues/16)

I have tried running it for 7 hours, but it still prints `OmniGraphSettings::getCudaDeviceOrdinal: unable to get a valid CUDA device id from the renderer. Defaulting to GPU0.` I suspected that it is still compiling shaders. 😭😭😭

cremebrule, 2024-12-09:
Hi @Snowflake-Pink ,

Hmmm I think this is the fundamental issue:

> ```
> 2024-12-06 08:41:43 [1,227ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
> 2024-12-06 08:41:43 [1,227ms] [Error] [carb.graphics-vulkan.plugin] vkEnumeratePhysicalDevices failed. No physical device is found.
> 2024-12-06 08:41:43 [1,227ms] [Error] [carb.graphics-vulkan.plugin] No physical device is found.
> 2024-12-06 08:41:43 [1,228ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
> ```

Seems like the GPU wasn't found for some reason. Can you verify that CUDA can be used in the python process, e.g.: by importing torch and running `torch.cuda.is_available()`?

Snowflake-Pink, 2024-12-10:
Hi @cremebrule ,

I am excited to announce that the solution to this issue!! Thanks a lot to the engineer from [AutoDL](https://www.autodl.com/)!! 🥳🥳

**This is probably that `libegl1` or other dependencies about Vulkan are not installed (especially on cloud server environment).**

Run Command:
```
apt update && apt install -y libegl1

OR

apt update && apt install -y vulkan-tools libvulkan1 libsm6 libglu1-mesa-dev libegl1
```

To solve. 🥳🥳

Thanks again to the engineers from [AutoDL](https://www.autodl.com/).

Snowflake-Pink, 2024-12-10:
This issue will be closed in a few days if there are not other comments.

Snowflake-Pink, 2024-12-11:
@cremebrule I greatly appreciate your kind help. This issue is closed.

closed 2024-12-11 by Snowflake-Pink