StanfordVL/BEHAVIOR-1K #1059: Inconsistency detected by [ld.so](http://ld.so/): ../elf/dl-tls.c: 517: _dl_allocate_tls_init: Assertion `listp != NULL' failed!

1725917163, 2024-12-19:
python -m omnigibson.examples.robots.robot_control_example --quickstart 
[INFO] [omnigibson.simulator] ----- Starting OmniGibson. This will take 10-30 seconds... -----
Starting kit application with the following args:  ['/home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/apps/omnigibson_4_1_0.kit', '--/app/tokens/exe-path=/home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['--quickstart']
[Info] [carb] Logging to file: /home/andy1/anaconda3/envs/omnigibson/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.1/kit_20241219_095118.log
2024-12-19 01:51:18 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.047s] [ext: omni.kit.async_engine-0.0.0] startup
[DEBUG] [asyncio] Using selector: EpollSelector
[0.114s] [ext: omni.stats-1.0.1] startup
[0.114s] [ext: omni.client-1.1.0] startup
[0.141s] [ext: omni.datastore-0.0.0] startup
[0.142s] [ext: omni.blobkey-1.1.0] startup
[0.142s] [ext: omni.ujitso.default-1.0.0] startup
[0.142s] [ext: omni.hsscclient-0.0.0] startup
[0.143s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.143s] [ext: omni.assets.plugins-0.0.0] startup
[0.145s] [ext: omni.gpu_foundation-0.0.0] startup
[0.151s] [ext: carb.windowing.plugins-1.0.0] startup
[0.159s] [ext: omni.kit.renderer.init-0.0.0] startup
Inconsistency detected by [ld.so](http://ld.so/): ../elf/dl-tls.c: 517: _dl_allocate_tls_init: Assertion `listp != NULL' failed!

cgokmen, 2024-12-19:
Question, what happens if you retry running a few times? We have sporadically run into this issue but it's not persistent.

hairuoliu1, 2024-12-22:
the same problem

hairuoliu1, 2024-12-23:
This problem was first reported in December 2015. The problem was caused by a bug in glibc, and it was completely fixed in the 2.34 version of glibc in August 2021. However, this version of glibc is currently only installed by default on RHEL 9 beta and Ubuntu 21.10 ( Naughty Indri)