StanfordVL/BEHAVIOR-1K #1074: python tests/test_data_collection.py

1725917163, 2025-01-07:
Traceback (most recent call last):
  File "/home/andy/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.ui/omni/physxui/scripts/extension.py", line 82, in on_stage_update
    cam_pos = camera.position_world
  File "/home/andy/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.viewport.utility/omni/kit/viewport/utility/camera_state.py", line 100, in position_world
    return self.usd_camera.ComputeLocalToWorldTransform(self.__time).Transform(Gf.Vec3d(0, 0, 0))
  File "/home/andy/anaconda3/envs/omnigibson/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.viewport.utility/omni/kit/viewport/utility/camera_state.py", line 87, in usd_camera
    raise RuntimeError(f'"{self.__camera_path}" is not a valid Usd.Prim or UsdGeom.Camera')
RuntimeError: "/World/viewer_camera" is not a valid Usd.Prim or UsdGeom.Camera

ChengshuLi, 2025-01-07:
This error is from Isaac and it's safe to ignore.

closed 2025-01-07 by ChengshuLi

Walkism, 2025-05-13:
I encountered the same problem, and more:

> Traceback (most recent call last):
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.widget.viewport/omni/kit/widget/viewport/impl/texture.py", line 355, in on_drawable_changed
    self.__drawable_changed_fn(texture['rp_resource'], event.payload.get('presentation_key', 0))
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.widget.viewport/omni/kit/widget/viewport/widget.py", line 465, in __set_image_data
    vp_api._sync_viewport_api(camera, canvas_size, vp_api.time, view, projection, force_update=True)
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.widget.viewport/omni/kit/widget/viewport/api.py", line 544, in _sync_viewport_api
    self.__projection = self._conform_projection(policy, camera, image_aspect, canvas_aspect, projection)
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.widget.viewport/omni/kit/widget/viewport/api.py", line 473, in _conform_projection
    gf_camera = camera.GetCamera(self.__time)
AttributeError: 'NoneType' object has no attribute 'GetCamera'
Traceback (most recent call last):
  File "/home/asus/isaacsim-4.1.0/extsPhysics/omni.physx.ui/omni/physxui/scripts/extension.py", line 82, in on_stage_update
    cam_pos = camera.position_world
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.viewport.utility/omni/kit/viewport/utility/camera_state.py", line 100, in position_world
    return self.usd_camera.ComputeLocalToWorldTransform(self.__time).Transform(Gf.Vec3d(0, 0, 0))
  File "/home/asus/isaacsim-4.1.0/kit/exts/omni.kit.viewport.utility/omni/kit/viewport/utility/camera_state.py", line 87, in usd_camera
    raise RuntimeError(f'"{self.__camera_path}" is not a valid Usd.Prim or UsdGeom.Camera')
RuntimeError: "/World/viewer_camera" is not a valid Usd.Prim or UsdGeom.Camera
> 
I followed [installation guidance](https://behavior.stanford.edu/omnigibson/getting_started/installation.html) strictly. Unfortunately, these errors occur in **local installation** as well as the **official container**.
May I ask how to "**ignore**" these errors? I get this error in almost every example in the [omnigibson official doc](https://behavior.stanford.edu/omnigibson/getting_started/examples.html), and it crashes the program everytime. 

(In addition, is isaac sim 4.1.0 still the best compatible version for the latest omnigibson? Or should users try the latest isaac sim 4.5.0 for less isaac sim bugs?)