StanfordVL/BEHAVIOR-1K #1825: Segmentation fault (core dumped)

hairuoliu1, 2025-09-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. python -m omnigibson.examples.robots.robot_control_example --quickstart

**Screenshots**
> (behavior) hairuo@hairuo-B660M-D2H-DDR4:~/Projects/BEHAVIOR-1K$ python -m omnigibson.examples.robots.robot_control_example --quickstart
> [00:00:02.885] [INFO] [omnigibson.simulator] ----- Starting OmniGibson. This will take 10-30 seconds... -----
> Starting kit application with the following args:  ['/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py', '/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/apps/omnigibson_4_5_0.kit', '--/app/tokens/exe-path=/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--/app/installSignalHandlers=0', '--ext-folder', '/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/apps', '--/physics/cudaDevice=0', '--portable']
> Passing the following args to the base kit application:  ['--quickstart']
> [Info] [carb] Logging to file: /home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/logs/Kit/OmniGibson/3.7/kit_20250913_225015.log
> 2025-09-13 14:50:15 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
> [0.031s] [ext: omni.kit.async_engine-0.0.1] startup
> [00:00:03.035] [DEBUG] [asyncio] Using selector: EpollSelector
> [0.134s] [ext: omni.metrics.core-0.0.1] startup
> [0.134s] [ext: omni.client.lib-1.0.0] startup
> [0.151s] [ext: omni.blobkey-1.1.2] startup
> [0.152s] [ext: omni.stats-1.0.1] startup
> [0.152s] [ext: omni.datastore-0.0.0] startup
> [0.156s] [ext: omni.client-1.2.2] startup
> [0.159s] [ext: omni.ujitso.default-1.0.0] startup
> [0.161s] [ext: omni.hsscclient-1.1.1] startup
> 2025-09-13 14:50:15 [153ms] [Warning] [omni.datastore] OmniHub is inaccessible
> [0.163s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
> [0.164s] [ext: omni.assets.plugins-0.0.0] startup
> [0.165s] [ext: omni.gpu_foundation-0.0.0] startup
> [0.170s] [ext: carb.windowing.plugins-1.0.0] startup
> [0.174s] [ext: omni.kit.renderer.init-0.0.0] startup
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 12 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 13 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 14 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 15 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 16 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 17 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 18 belongs to.
> 2025-09-13 14:50:15 [195ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 19 belongs to.
> 
> |---------------------------------------------------------------------------------------------|
> | Driver Version: 575.57.08     | Graphics API: Vulkan
> |=============================================================================================|
> | GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
> |     |                                  |        |     |            | Device-ID | UUID       |
> |     |                                  |        |     |            | Bus-ID    |            |
> |---------------------------------------------------------------------------------------------|
> | 0   | NVIDIA GeForce RTX 3070 Ti       | Yes: 0 |     | 8438    MB | 10de      | 0          |
> |     |                                  |        |     |            | 2482      | dcad02d6.. |
> |     |                                  |        |     |            | 1         |            |
> |=============================================================================================|
> | OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-139-generic
> | XServer Vendor: The [X.Org](http://x.org/) Foundation, XServer Version: 12013000 ([1.20.13.0](http://1.20.13.0/))
> | Processor: 12th Gen Intel(R) Core(TM) i7-12700KF
> | Cores: 12 | Logical Cores: 12
> |---------------------------------------------------------------------------------------------|
> | Total Memory (MB): 31927 | Free Memory: 23743
> | Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
> |---------------------------------------------------------------------------------------------|
> [0.822s] [ext: omni.kit.pipapi-0.0.0] startup
> [0.823s] [ext: omni.kit.pip_archive-0.0.0] startup
> [0.823s] [ext: omni.kit.telemetry-0.5.1] startup
> [xcb] Unknown sequence number while appending request
> [xcb] You called XInitThreads, this is not your fault
> [xcb] Aborting, sorry about that.
> [xcb] Extra reply data still left in queue
> [xcb] This is most likely caused by a broken X extension library
> [xcb] Aborting, sorry about that.
> Fatal Python error: Segmentation fault
> 
> Thread 0x00007f5d91775640 (most recent call first):
[... 27 lines cut ...]
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
>   File "PythonExtension.cpp::startup()", line 2 in <module>
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 427 in _start_app
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 212 in __init__
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/simulator.py", line 147 in _launch_app
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/simulator.py", line 258 in _launch_simulator
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/envs/env_base.py", line 103 in __init__
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/utils/python_utils.py", line 92 in wrapper
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 101 in main
>   File "/home/hairuo/Projects/BEHAVIOR-1K/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 189 in <module>
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/runpy.py", line 86 in _run_code
>   File "/home/hairuo/miniconda3/envs/behavior/lib/python3.10/runpy.py", line 196 in _run_module_as_main
> 
> Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._dynamo.autograd_compiler, torch._C._dynamo.eval_frame, torch._C._dynamo.guards, torch._C._dynamo.utils, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._sp [... cut ...] _filter_1d, _ni_label, scipy.ndimage._ni_label, scipy.special.cython_special, scipy.stats._stats, scipy.stats._sobol, scipy.stats._qmc_cy, scipy.stats._biasedurn, scipy.stats._stats_pythran, scipy.stats._levy_stable.levyst, scipy.stats._ansari_swilk_statistics, scipy.stats._mvn, scipy.stats._rcont.rcont (total: 249)
> Segmentation fault (core dumped)

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Isaac Sim Version 4.5.0
 - OmniGibson Version latest
 - GPU 3070Ti

DarthAuto, 2025-10-07:
Hi @hairuoliu1, I encountered exactly the same issue. Is it solved by now?

starshine360, 2025-11-15:
Hi, @hairuoliu1  I encountered the same issue. Could you please let me know if you know how to resolve it?

wensi-ai, 2025-11-18:
Hi, this is a known issue with Ubuntu 20.04, could you try

```
conda remove libxcb xorg-libx11
```

And see if it solve the issue? If it's still not working, you can try reinstall xorg-libx11 to match the local libx11 version on your machine.

Also, if you don't need evaluation you can also try to setup behavior without --eval flag.