StanfordVL/BEHAVIOR-1K #1832: RLDS builder not found

YandongJi, 2025-09-18:
**Describe the bug**
When accessing RLDS builder for OpenVLA-OFT using the link in Data conversion, having 404 not found error. Specifically it it this link: https://behavior.stanford.edu/https:/raw.githubusercontent.com/StanfordVL/b1k-baselines/main/tutorials/RLDS_builder/README.md

**Screenshots**
[image]

**Desktop (please complete the following information):**
N/A

**Additional context**
N/A

YandongJi, 2025-09-18:
Found it [here](https://github.com/evansh666/openvla-oft/blob/0632686798df7a5fad6885ab392bf3f0acab08c9/RLDS_builder/behavior_dataset/behavior_turn_on_radio/behavior_turn_on_radio_dataset_builder.py)

closed 2025-09-18 by YandongJi

guoyanzi, 2025-10-17:
cd RLDS_builder
tfds build --data_dir /path/to/save/rlds/dataset 
报错
FileNotFoundError: Could not find dataset generation script: /home/yanjie.guo/workspace/b1k-baselines/baselines/openvla-oft/RLDS_builder/RLDS_builder.py. 
进入到这个文件夹 RLDS_builder/behavior_dataset/behavior_turn_on_radio：
再运行，报错
 from RLDS_builder.behavior_dataset.utils.conversion_utils import MultiThreadedDatasetBuilder 
ModuleNotFoundError: No module named 'RLDS_builder'